n0xa/m5stick-nemo #10: Insufficient program storage space on a M5Stick C device

vladimirpetrov, 2023-09-29:
Hi, I'm trying this code on a M5Stick C device and I'm getting the following:

Sketch uses 1418390 bytes (108%) of program storage space. Maximum is 1310720 bytes.
Global variables use 63076 bytes (19%) of dynamic memory, leaving 264604 bytes for local variables. Maximum is 327680 bytes.

I'm using Arduino IDE 2.2.1 with libraries M5StickC ver. 0.2.9 M5StickCPlus ver. 0.1.0 and M5Unified ver. 0.1.9 installed.
I have the same result with Arduino 1.8.19.

n0xa, 2023-09-30:
Hi! In Arduino IDE, you must switch partition schemes. 
Tools -> Partition Scheme -> No OTA (Large APP)

I will add this to the readme soon.

vladimirpetrov, 2023-10-01:
This helped, thanks!

closed 2023-10-01 by vladimirpetrov