n0xa/m5stick-nemo #17: Add wifi deauther

ashishvala15, 2023-11-12:
Add more wifi attack like deauth

zach67385, 2023-11-12:
would be cool to se

Manthany2000, 2023-11-13:
yes! its the only reason I switch between the 2. This has a better sour apple compaired to Marauder. PLZZZZZZZ ADDDD

n0xa, 2023-11-13:
In the USA, per section 333 of The Communications Act of 1934, deauthentication attacks against a network you are not an authorized administrator of, are 100% illegal. This includes targeted deauthentication of a network that's not yours, and, more severely, bulk "deauthentication spam" type attacks such as those commonly performed by users of DSTIKE, Marauder, and Flipper Dev Board users. 

Colloquially, FCC regulations, CFR Title 47 Part 15.3(m) defines "harmful interference" as `Any emission, radiation or induction that endangers the functioning of a radio navigation service or of other safety services or seriously degrades, obstructs or repeatedly interrupts a radiocommunications service operating in accordance with this chapter. `

Which is to say that the FCC categorizes unauthorized deauthentication in the same way it categorizes GPS and Cell Phone Jamming, and they have penalized the most egregious perpetrators accordingly. In the most heavy-handed punishment the FCC Fined an organization $718,000 USD for repeatedly knocking users off of nearby WiFi networks that they did not own. 

That is to say, I do not think of Deauthentication Attacks as a mere prank. It is often abused by people who have no idea the problems they could be causing, and crosses the line into malicious intent. I would further urge anyone who uses the Deauthentication tools mentioned, to think twice before abusing them against unauthorized networks. It is not "hacking."

ashishvala15, 2023-11-14:
Complete

closed 2023-11-14 by ashishvala15