n0xa/m5stick-nemo #22: Feature Request

grdashark, 2023-11-21:
Hi mate, can you add an android ble spam feature or Google Fast Pair Service spammer? Thanks!

n0xa, 2023-11-22:
I found the flipper Xtreme Firmware Implementation, but it's fairly bloated for this project. I'll have to figure out how this protocol works at a low level and then work to boil it down to something that is more appropriate for the minimal interface I've got here.  I'll add it to the list of potential features.

IMXNOOBX, 2023-11-23:
Well, it isnt too diferent from the the other ble spams, just changing the brand and some other tweaks in the packets will work. You can check my implementation of it below, if you want i can try to implement it nemo.

https://github.com/IMXNOOBX/fstick/blob/b1d38a5a348369a4c65aac78a84be8c64587796d/classes/ble.h#L185-L210

I8Pizza2Day, 2023-12-01:
Love Nemo for the M5StickC Plus, was curious if you could add the Wifi Sniffer tools of the ESP32Marauder? Keep up the Awesome work! Just starting to learn more myself. But wanting the Deauth Sniffer tools for a project in a Clean GUI like your NEMO install.

n0xa, 2023-12-28:
This has been addressed with Pull Request #35

closed 2023-12-28 by n0xa