n0xa/m5stick-nemo #24: Cardputer menu navigation

ludothegreat, 2023-11-28:
I'm having an issue with the menu on my Cardputer in version 2.0.1. When selecting any item in the menu, both with the > button or the 'ok' button it seems to be doubling the input and selecting the first menu item in the next screen.

Example: If I select any item from the main menu it will enter the sub menu for a second and then go back. 

It'll randomly work on occasion or it might be that I can sometimes spam the down button immediately after pressing > or 'ok' to get it to work. It's one out of like 10 times, though.

n0xa, 2023-12-01:
Yes, I think there might need to be a longer delay on the inputs or it might be polling the keyboard too often. the keyboard on this thing is an odd case and the buttons are not handled at all like the hardware buttons on the M5Sticks. On the Cardputer, I just very quickly tap the enter button. I agree it's a bit flaky still. PRs welcome, but perhaps wrap an additional delay in an `#if defined CARDPUTER` to avoid lagging the other devices?

facelesstech, 2023-12-02:
Thanks for replying, I'm very new to using the M5 stack so still trying to get my head round the code. I'm more used to just using arduino's or the odd esp8266. This is an amazing project BTW, I want to add a IR Remote to this project but I noticed that the IR led is on pin 44 on the cardputer and pin 9 on the m5stack, I need to find away to have it change to the right pin of the right device in the define section. I just changed it in the  tvbg.h file for now.

My very ambitious plan would be to use the cardputers sd card to hold the IRDB from the flipper and use those codes because that's all I used to use my flipper for before I sold it. I have no idea if this is even possible and how long it would take me to pull off but it would make this project a very powerful contender to the flipper.

ludothegreat, 2023-12-02:
The quick tap works. Thanks for the tip! It is still a bit flaky, but it's much more usable for now. Thanks for the work you've put in so far. This is a very cool project.

n0xa, 2023-12-20:
This was fixed by #26 -- Thanks, @facelesstech !

closed 2023-12-20 by n0xa