n0xa/m5stick-nemo #28: Feature requesr

Pufferfish1223, 2023-12-17:
Is there any way to make the ir pin one of the gpios so I can use a more powerfull ir led. Thanks

n0xa, 2023-12-20:
I moved the IR LED pin number to the DEFINE blocks so it should be easier to compile it for a custom pin. You may need to play with the ACTIVE_LOW_IR option, some external LED attachments need that defined, others need it removed. Your most likely option is to use pin G26 on the M5Stick C Plus. If you have very good results increasing the distance with a specific add-on LED, let me know which module you're using. If it's popular enough, I can add a setting option for external LED boards if we have a known one that works well. I can't invest effort into that until I know it works really well.

Pufferfish1223, 2023-12-20:
It worked, thank you so much, the range is now atleast 10 meters just using a generic ir led salvaged from a old remote.

I8Pizza2Day, 2023-12-20:
@Pufferfish1223 How do you have the ir Led attached? any pics or drawings?

Pufferfish1223, 2023-12-21:
The negative lead of the ir led is plugged in to the GND port and the positive is plugged into the G26 port of the m5stickc plus.
[image]

n0xa, 2023-12-27:
I will likely make this more easily configurable in an upcoming release.

closed 2023-12-27 by n0xa