n0xa/m5stick-nemo #31: Please consider using the 433Mhz receive and transmitter from M5Stack

shadowdoggie, 2023-12-23:
Hello n0xa :)

I highly admire what you're doing, and i am really hoping you could take a look at some of the modules/hats, i found for the m5stick, that you can connect all at the same time, using a 1 to 3 hub expansion unit, that they also sell here: https://shop.m5stack.com/products/mini-hub-module

Here are the 433 Mhz modules:
Receiver: https://shop.m5stack.com/products/rf-unit-433mhz-receiver-syn513r
Transmitter: https://shop.m5stack.com/products/rf-unit-433mhz-transmitter-syn115
IR Receiver + Transmitter: https://shop.m5stack.com/products/ir-unit

p.s.
Maybe they have some more interesting modules/hats?
I don't have these modules myself sadly (yet).

Thank you very much in advance!

shadowdoggie, 2023-12-23:
They also sell a RFID module :) https://shop.m5stack.com/products/rfid-unit-2-ws1850s

n0xa, 2023-12-27:
If someone wants to donate this hardware to me, I will see if I can do something with it, but there's not a good way to store the received signals, and to replay you would have to unplug the receiver and then attach the transmitter. These features are running up against the practical use-cases for the M5StickCPlus hardware platform.

tranquocnhat94, 2024-03-14:
https://www.youtube.com/watch?v=Q7fftVPiFSE&t=198s

I think it would be interesting if we combined it with an MQTT server, or google sheets, you can save everything on there and retrieve them when you need to use them, the 433 modules are quite compact, this would be an idea. Great idea as an alternative to the expensive flipper zero

tranquocnhat94, 2024-03-14:
https://vi.aliexpress.com/item/1005003328002232.html?gatewayAdapt=glo2vnm

This module is very cheap, maybe I can sponsor you a more worthy amount

Awlexegrecki, 2024-03-21:
@n0xa I found firmware on the M5Burner for identifying tags with the RFID 2 scanner:
`#include <M5StickCPlus2.h>
#include <Wire.h>
#include "MFRC522_I2C.h"

#define MFRC522_I2C_ADDRESS 0x28
#define RST_PIN -1

MFRC522_I2C mfrc522(MFRC522_I2C_ADDRESS, RST_PIN, &Wire);

void setup() {
  M5.begin();
  Wire.begin();
  mfrc522.PCD_Init();
  M5.Lcd.setRotation(1);
  M5.Lcd.fillScreen(BLACK);
  M5.Lcd.setTextSize(2);
  M5.Lcd.setCursor(0, 0);
  M5.Lcd.println("RFID Reader");
  M5.Lcd.println("Approchez un badge");
}

void loop() {
  if (mfrc522.PICC_IsNewCardPresent() && mfrc522.PICC_ReadCardSerial()) {
    M5.Lcd.fillScreen(BLACK);
    M5.Lcd.setCursor(0, 0);
    M5.Lcd.setTextColor(WHITE);
    M5.Lcd.print("UID: ");

    String currentUID = "";
    for (byte i = 0; i < mfrc522.uid.size; i++) {
      if (mfrc522.uid.uidByte[i] < 0x10) {
        currentUID += "0";
      }
      currentUID += String(mfrc522.uid.uidByte[i], HEX);
      M5.Lcd.print(currentUID.substring(i * 2, i * 2 + 2));
      if (i != mfrc522.uid.size - 1) {
        M5.Lcd.print(" ");
      }
    }

    M5.Lcd.setCursor(0, 20);
    M5.Lcd.print("Type: ");
    String cardType = mfrc522.PICC_GetTypeName(mfrc522.PICC_GetType(mfrc522.uid.sak));
    M5.Lcd.println(cardType);

    M5.Lcd.setCursor(0, 40);
    M5.Lcd.print("Taille: ");
    byte cardSize = mfrc522.uid.size;
    M5.Lcd.print(cardSize);
    M5.Lcd.println(" octets");

    delay(2000);
    mfrc522.PICC_HaltA();
    M5.Lcd.fillScreen(BLACK);
    M5.Lcd.setCursor(0, 0);
    M5.Lcd.println("Approchez un badge");
  }
}`

Havekt723, 2024-05-08:
So did you guys get it working? The sub gigahertz

closed 2024-08-22 by n0xa