n0xa/m5stick-nemo #32: Wi-Fi Deauther and Other Feature request

shadowsss, 2023-12-25:
Could you add new feature like this https://github.com/risinek/esp32-wifi-penetration-tool

shadowdoggie, 2023-12-26:
You should take a look at this firmware: https://github.com/IMXNOOBX/fstick
It has all you're asking for :)

n0xa, 2023-12-27:
I've closed a few feature requests for Deauthentication. I will never add Deauth to NEMO. It falls into the category of illegal harmful interference by the FCC here in the USA - and is punishable just the same as if you were jamming GPS, police radios or mobile phones. You can get your kicks using other peoples' firmware. It's not a prank, and it's not in the spirit of what I want NEMO to enable people do learn about.

closed 2023-12-27 by n0xa