n0xa/m5stick-nemo #34: WIFI and Bluetooth attacks broken in 2.0.1 and 2.0.3

bicurico, 2023-12-26:
Hello,

On a new M5StickC+ versions 2.0.2 and 2.0.3 don'wtoutput WIFI or Bluetooth.

For example, activating Rickroll, nothing happens.

When flashing Marauder or Nemo v1.4.0, everything works as expected.

bicurico, 2023-12-26:
I used M5Burner, selected STICKC on the left menu and then searched "nemo".
From the list I selected "NEMO For M5Stick C Plus". WiFi Spam and Bluetooth device pairing spam is not working on 2.0.1 and 2.0.3.
WIth 1.4.0 it works.

n0xa, 2023-12-27:
I don'y know why it's not working for you. It makes bluetooth spam pop-ups on my wife's Windows 11 laptop, on my work iPhone and personal iPad. I can also see the Wifi spam access points in my mobile devices wifi lists -- though, I have never seen the wifi show up on Windows wifi selector, even with Marauder or other spamming tools.

bicurico, 2023-12-27:
> Não costuma receber e-mails de ***@***.*** Saiba por que motivo isto é importante<https://aka.ms/LearnAboutSenderIdentification>

I was hoping that there was just an error with the M5StickC+ and the Cardputer versions uploaded.

Enviado de Outlook para Android<https://aka.ms/AAb9ysg>

bicurico, 2023-12-27:
BT seems to be working, afterall, but WIFI not.
Because it does work with v1.4.0, I compiled it myself and replaced beaconSpam() and beaconSpamList() from 2.0.3 with beaconSpamList() from 3 month ago.
Now it is working.
I did not have time to check what the difference is and why it made a difference.

n0xa, 2023-12-28:
I just published 2.1.0 and I installed it from M5Burner onto my StickC-Plus. All spamming is working fine for my devices. What device are you using to view WiFi Spam?

bicurico, 2023-12-29:
I just uploaded 2.1.0 and it does not work with WIFI.

I am using a Redmi Note 11 Pro (Android). With "my" build (as described above) I see i.e. all Rickroll AP's after 5 seconds. With your build nothing shows up.

I tested with my PC (Windows 11) and it's builtin BT. The same happens. With my build I see the spamming.

Not really sure what the problem might be.

n0xa, 2023-12-29:
This sounds like the same problem I had after 1.4.0 that I thought I'd completely resolved with the 2.x builds. 1.5.x builds were not showing wifi spam on many devices. Mostly, I was able to speed up the main program loop a bit and that seemed to have resolved it for my use cases and complaints have been much fewer since the 2.0 release. If you can put your fork in your GitHub, I'll see if I can work your changes back into the main branch. I can't quite recall why i split the beaconSpam up the way I did.

bicurico, 2023-12-30:
Hi,

I just put my fork on GitHub. Hope it helps.

Sorry for not being more collaborative, but I am new to GitHub, Arduino SDK, C and M5 stuff.

Regards,
Vitor

n0xa, 2023-12-30:
I'm also pretty new to all of this. We can sort it out together!

closed 2024-02-02 by n0xa