n0xa/m5stick-nemo #37: Issues with 2.1.0 on cardputer

geekonthebeach, 2023-12-29:
Tried loading a few times on my new cardputer and the screen just will not stay on. It'll flash if I press the top reset button, but just turns back off. Reverted back to 2.0.3 and is all stable.

n0xa, 2023-12-29:
When you get a chance, can you try installing 2.1.0 again, and from either M5Burner or Arduino IDE, activate the serial monitor. In M5Burner, it's under the 3-line menu in the top right. Click "connect" and then click the "reboot" button on the monitor. The serial monitor should spit out EEPROM values. Can you paste those for me here?

n0xa, 2023-12-29:
I also just added a quick fix to the main branch and on M5Burner, I published a build 2.1.3. With this, if you boot it up while holding the "next" key (that is, the down arrow or tab button on the cardputer) it will wipe the first four bytes of EEPROM, and when you let go of the button, it'll initialize those bytes with default values including full brightness on the LCD screen. See if that build works and can fix your problem.

geekonthebeach, 2023-12-29:
Using M5Burner.  Here are the EEPROM values while running 2.1.0:

EEPROM 0: 1
EEPROM 1: 15
EEPROM 2: 0
EEPROM 3: 0

Not seeing 2.1.3 available in M5Burner yet.

n0xa, 2023-12-29:
I uploaded 2.1.3 but didn't hit the "publish" button - oops. It should be in there now. 

EEPROM 0 is rotation
EEPROM 1 is the backlight dimmer time in seconds
EEPROM 2 is brightness and 0 is actually PWM - 155 out of 255 - on my Cardputer, I can still make the screen out in a darkened room at this level.  But that's probably the issue.
EEPROM 3 is the TVBGone region of Americas/Asia. 

I'm pretty sure that resetting the EEPROM with the down-arrow on boot with the new Cardputer version will fix it. Let me know!

geekonthebeach, 2023-12-29:
I tried the EEPROM reset with the down arrow several times and nothing seemed to happen.  I just saw 2.1.3 in M5Burner and burned it to the cardputer.  Seems to be all good now.  Thanks!

n0xa, 2023-12-29:
The EEPROM Reset was just added in 2.1.3, specifically to help you reset your EEPROM (but it's a nice thing to have for troubleshooting). Glad you've got it working now!

closed 2023-12-29 by n0xa