n0xa/m5stick-nemo #42: M5StickCPlus2 Screen very dim on 2.2.1 firmware

CtrlAltDel-Irl, 2023-12-30:
2.2.1 flashes fine.  At first i thought it was in a bootloop or not booting but then noticed very dim lettering on the screen.
Can't test further as i can't read it.

Looking forward to testing the next release.

n0xa, 2023-12-31:
You can reset the eeprom with "Clear Settings" from the settings menu if you can see it, or by holding tab or down-arrow while powering on. This should enable full brightness regardless what was previously in eeprom causing the issue.

CtrlAltDel-Irl, 2023-12-31:
Thanks very much n0xa.  I'll try flashing again in the morning and will report back.
Again, many thanks :)

n0xa, 2024-01-28:
I finally got my Plus2 in this week and calibrated the backlight minimum and maximum brightness for the new platform as part of #81. Should be better in v2.4.1

closed 2024-01-28 by n0xa