n0xa/m5stick-nemo #44: Make it so you can change the ir led pin in settings

SuperSeno, 2024-01-01:
Could you please make it so you can change the IR pin ether 9 or 26 in settings?

FNTTRTXM5, 2024-01-30:
On G26 an 5out

n0xa, 2024-02-02:
I haven't forgotten about this. It might be a good project for someone who wants to take a crack at contributing.  A few things I've been thinking about:

LED pin should be able to be set to Pin 26, 32, or 33 on StickC, Plus and Plus2.  32/33 are on the GROVE port next to the USB port. On Cardputer, Pin 1 or 2 are exposed on the GROVE port.

The settings must also allow for active-high (digitalWrite HIGH turns on the LED) or active-low (digitalWrite LOW turns on the LED). The StickC Plus and Plus2 use Active-low with the internal LED for example, and some external IR LEDs are also active-low, particularly high-power ones that use the 5VDC pin. 

I hope to get around to this eventually if someone else doesn't. implementing it by itself isn't hard, but implmenting it cleanly might be a little challenging.

FNTTRTXM5, 2024-02-03:
Here is an example:
/// TV-B-GONE ///
void tvbgone_setup() {
  M5.dis.clear();

  pinMode(5, OUTPUT);   //Pin 5 (5OUT)
  pinMode(26, OUTPUT);  // Pin 26

  // ...

  digitalWrite(5, HIGH);  // Schalte IR-LED an Pin 5 ein
  digitalWrite(26, HIGH); // Schalte IR-LED an G26 ein

  delay_ten_us(5000);

  if (region == NA) {
    M5.dis.println(TXT_RG_AMERICAS);
  } else {
    M5.dis.println(TXT_RG_EMEA);
  }

  M5.dis.println(TXT_SEL_GO_PAUSE);
  M5.dis.println(TXT_SEL_EXIT);
  delay(1000);
}

closed 2024-08-22 by n0xa