n0xa/m5stick-nemo #48: Add more functionality to ir remote

mceres, 2024-01-04:
Here is a repository with a big DB of remotes, might help.

https://github.com/UberGuidoZ/Flipper-IRDB

n0xa, 2024-01-06:
This is pretty neat, but it's not handled in the same format as the Arduino TV-B-Gone code. If someone is looking to help with NEMO, coming up with a way to either directly parse flipper zero IR files, or to convert a batch of them into the array structures already used in NEMO would be appreciated. I'm personally not certain where to start on it.

mceres, 2024-01-07:
Does this help?

https://github.com/crankyoldgit/IRremoteESP8266

n0xa, 2024-01-08:
I'm already using that library. What I mean is I would like someone to help contribute to integrating the new IR stuff into NEMO if possible. I'm in the middle of job hunting and focused more on wifi functionality at the moment.

closed 2024-04-19 by n0xa