n0xa/m5stick-nemo #56: TB-B-Gone don't work

Elitesniper06, 2024-01-09:
I have version 2.3.2-PT_BR of M5Burner and I have tried it with several televisions and it does not turn any of them on or off when it did before.
I'm from the US and I have the M5StickCPlus2.

n0xa, 2024-01-18:
I am waiting for my plus2 to arrive so I can once and for all make sure TV-B-Gone works. If you can compile the project, try uncommenting line 55 `//#define ACTIVE_LOW_IR` but this also seemed to cause a problem with the LED remaining lit all the time. I can solve that once I have a Plus2 of my own. I'm kind of flying blind on that platform right now.

n0xa, 2024-01-28:
I finally got my Plus2 and included this fix in #81 which merged into NEMO v2.4.1

closed 2024-01-28 by n0xa