n0xa/m5stick-nemo #61: Nemo FW 2.3.2 M5StickCPlus2  Screen Blacked Out

CtrlAltDel-Irl, 2024-01-11:
After flashing 2.3.2 the screen remains dark.  Pressing both side buttons momentarily resets the screen but then goes dark again.  Can't perform further testing due to this.

Looking forward to next release.

CtrlAltDel-Irl, 2024-01-11:
Reflashed a few times and once the screen was visible.  Upon rebooting the screen remained dark again.  Subsequent flashes remain dark.

CtrlAltDel-Irl, 2024-01-11:
Update:  Flashed to another M5StickCPlus2, screen was dim but legible.  Reset in the settings menu and all is good.
Looks like a hardware issue on the 1st stick as all other firmwares tried display the same behaviour.

n0xa, 2024-01-12:
If the screen is super-dim, especially on the StickC-Plus2 or the Cardputer, clear the settings.  The easiest way to do this on 2.2.1 or newer is to hold down the SELECT  (side) button on the StickC models, or the TAB or DOWN ARROW keys on the Cardputer. 

If you go somewhere very dark, you can probably see the screen to navigate to the Settings -> Clear Settings option menu. 

This happens because starting with 2.2.0, when brightness support was added to the two aforementioned platforms, somehow the byte of EEPROM settings storage was set to a very low value, likely 0. I haven't seen this in the wild myself. I could make the minimum brightness higher on these two boards, I suppose, but as it stands, I personally like how dim the Cardputer display is at the lowest setting, and I feel the StickC Plus is just a little too bright even when fully dim.

CtrlAltDel-Irl, 2024-01-12:
Thanks n0xa,

Yeah  it's defo a hardware issue on the M5StickCPlus2.  Closing this issue.

closed 2024-01-12 by CtrlAltDel-Irl