n0xa/m5stick-nemo #66: Hey, I found a way to display the bitmap in the M5StickCPlus

Niximkk, 2024-01-17:
The way i found was to make a monochrome bitmap using https://javl.github.io/image2cpp/ BUUUT using "Horizontal - 2 bytes per pixel(565)" as the draw mode. You also have to use the correct size ( change it to `BITMAP M5.Lcd.drawBitmap(0, 0, M5.Lcd.width(), M5.Lcd.height(), NEMOMatrix)`).

This 'fix' was found at https://community.m5stack.com/topic/2305/solved-displaying-images-various-formats/10.

[image]

Niximkk, 2024-01-17:
GG Found a way to use a colored bitmap. Just follow this guide.
https://www.youtube.com/watch?v=ahErgztMYVE

The program that he used in the video is this one: 
> https://sourceforge.net/projects/lcd-image-converter/

Here is the bitmap already converted (with colors):
> https://e-z.tools/p/bdwmuv9943

n0xa, 2024-01-18:
Is it possible you could try to work this into a functional pull request against the develop branch of the project? I think there's ample room in memory to add the bmp as another const with a slightly different name.

Niximkk, 2024-01-18:
I don't think that's a good idea, maybe in the future you will need that space. I was hoping you could test the bitmap on another device (since i only have the m5stickcplus) and if it works on all devices change the bitmap to this new one.

Niximkk, 2024-01-18:
I will fix the screen size tho.

closed 2024-01-30 by Niximkk