n0xa/m5stick-nemo #79: 'No serial data received' When Burning

rf-peixoto, 2024-01-27:
Straight to the point, when trying to burn the firmware on the device, the process does not establish communication. Step by step of what I'm doing, starting with the platforms:

- I'm on a Windows 11 22631.3085.
- The device is a cardputer in version M5Stack v0.9.

I tried following the tutorials listed in this repository. Starting with the M5 Burner:

1. Installed FTDI drivers.
2. Installed M5 Burner.
3. Downloaded the firmware in STAMPS3/Nemo. When clicking Burn I get the same error.

After a few attempts, I tried to perform the manual installation via ESPTool. I received the same error, which can be seen in [this screenshot](https://i.imgur.com/uXwmnPg.png).

On the [troubleshooting page](https://docs.espressif.com/projects/esptool/en/latest/esp32/troubleshooting.html) there is a paragraph about the error, saying that it could be a hardware problem (sh*t). It redirects to a [bootloader related page](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html#manual-bootloader). According to it, the loader should be selected automatically when connecting the device. There is a paragraph about executing the process manually, but it talks about "user guides" (my device came sealed, but without a manual(?)). The [user guide listed in the documentation](https://docs.m5stack.com/en/quick_start/Cardputer/arduino) does not say anything about this, it is just a guide for installing the libraries in the Arduino IDE.

Am I missing anything?

n0xa, 2024-01-28:
In device manager under ports (com and lpt) do you see your device show up? Any errors? You might also try connecting a wire from G0 to GND on the front panel. This will place the ESP32 inside into a bootloader mode that waits for esptool or M5Burner to upload something.

rf-peixoto, 2024-01-28:
Nothing in device manager, not even an error. I made sure to check the "Scan for hardware changes" option. In the Arduino IDE, however, I see that it recognizes it as UNKNOWN COM3. I searched whether it was possible to burn the binary there, [apparently not](https://forum.arduino.cc/t/use-ide-to-flash-a-pre-compiled-binary-hex/1132384).

By "front panel" do you mean the chipset pins, where there is the USB-C input? There are two GND there, but the only G0 I noticed (at least identified) was the button to the left of the ON/OFF Switch.

I ask before testing because I'm not very experienced with hardware, I'm afraid of making a wrong pin and damaging the device. For the record, I asked the same question on the [M5 Stack forum](https://community.m5stack.com/topic/6026/cardputer-com-not-working). If there is a solution there, I'll bring it here and vice versa.

Mmatuda, 2024-01-28:
Have you tried another cable? And other USB port? If it doesn't work, try pressing download button on StampS3, then plug in you PC, while pressing it. This button is hidden under the StampS3 sticker (right under that green ellipse), so you don't need to use wires.

rf-peixoto, 2024-01-28:
@Mmatuda I tried all available USB ports, even including a USB hub between the Cardputer and the notebook. I just tested this hidden button (I didn't even know it was there), but it didn't work either. On the M5 Stack forum they also recommended trying another cable. I will do this as soon as I purchase a new cable, I will post an update about that here.

rf-peixoto, 2024-01-29:
New cable, same error. In fact, I just tested it on Fedora 39 (fresh installed), I still haven't been able to put the device in download mode. Maybe I'm forgetting a drive, which makes the device "recognize" the computer as a charger. Still, I reinstalled them all again, with no result.

Mmatuda, 2024-01-30:
It seems like you got a faulty hardware... If you followed the instructions on M5Sack page, you shouldn't get that error.
Did you try contacting the seller?

rf-peixoto, 2024-01-30:
Yes, waiting for response. Also, tried all the solutions suggested on M5 forums, here's video to make sure I'm not crazy: https://www.youtube.com/watch?v=Vlx79AmbrPk Tried the M5 web burner too: https://uiflow2.m5stack.com/

rf-peixoto, 2024-03-02:
Solved. Bought some new cables, pressed BtnGo and connected. The S3 LED did not blink, but it worked. Thanks for your comments. For reference, the whole discussion about this issue on the M5 forums: https://community.m5stack.com/topic/6026/cardputer-com-not-working/

closed 2024-03-02 by rf-peixoto