n0xa/m5stick-nemo #96: Lags after using marauder

rioooxd, 2024-02-14:
I recently flashed esp32 marauder on my M5Stickc Plus2 and it kinda sucked so I switched back to Nemo which I have been using before and it has been working great but now it's lagging a lot. The clock is glitching with the menu and also I have to click the select button a few times for it to even respond. I think that marauder shit broke my m5stickc or something. What should I do? I tried flashing nemo again or using older versions and the same thing happens.

Niximkk, 2024-02-22:
Weird... try erasing the device. 
https://esp.huhn.me/

n0xa, 2024-02-23:
You can also erase using the M5Burner app. When you select to "burn" a firmware from M5Burner, there are 3 options: Cancel, Erase and Start. If you click the Erase option, it will completely wipe the device, then you can go back and burn the firmware again. I can confirm that this sometimes helps if the device is acting strange.

closed 2024-02-23 by n0xa