n0xa/m5stick-nemo #100: Error compiling with arduino app/cli

sukys, 2024-02-17:
Hello there.

Trying to upload to a M5 Cardputer and got an error (bellow is the error output from arduino-cli)
I Tried with arduino app (both versions 2.3.1 and 1.8) and also with arduino-cli, always following the read.me instructions.
Is tehere anything else I can do? 

```
$ arduino-cli compile --fqbn m5stack:esp32:m5stack_cardputer -e --build-property build.partitions=huge_app --build-property upload.maximum_size=3145728 ./m5stick-nemo.ino
/home/[MY_USER]/.arduino15/packages/m5stack/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /home/[MY_USER]/.arduino15/packages/m5stack/hardware/esp32/2.1.0/tools/sdk/esp32s3/lib/libnet80211.a(ieee80211_output.o): in function `ieee80211_raw_frame_sanity_check':
(.text.ieee80211_raw_frame_sanity_check+0x5c): multiple definition of `ieee80211_raw_frame_sanity_check'; /tmp/arduino/sketches/1F11C807588B4D5C4332A895BDD82DE1/sketch/m5stick-nemo.ino.cpp.o:/home/[MY_USER]/workspace/m5stack/cardputer/m5stick-nemo/deauth.h:19: first defined here
collect2: error: ld returned 1 exit status

Used library      Version Path
M5Cardputer       1.0.2   /home/[MY_USER]/Arduino/libraries/M5Cardputer
M5Unified         0.1.12  /home/[MY_USER]/Arduino/libraries/M5Unified
M5GFX             0.1.12  /home/[MY_USER]/Arduino/libraries/M5GFX
EEPROM            2.0.0   /home/[MY_USER]/.arduino15/packages/m5stack/hardware/esp32/2.1.0/libraries/EEPROM
IRremoteESP8266   2.8.6   /home/[MY_USER]/Arduino/libraries/IRremoteESP8266
DNSServer         2.0.0   /home/[MY_USER]/.arduino15/packages/m5stack/hardware/esp32/2.1.0/libraries/DNSServer
WiFi              2.0.0   /home/[MY_USER]/.arduino15/packages/m5stack/hardware/esp32/2.1.0/libraries/WiFi
WebServer         2.0.0   /home/[MY_USER]/.arduino15/packages/m5stack/hardware/esp32/2.1.0/libraries/WebServer
ESP32 BLE Arduino 2.0.0   /home/[MY_USER]/.arduino15/packages/m5stack/hardware/esp32/2.1.0/libraries/BLE
FS                2.0.0   /home/[MY_USER]/.arduino15/packages/m5stack/hardware/esp32/2.1.0/libraries/FS
SD                2.0.0   /home/[MY_USER]/.arduino15/packages/m5stack/hardware/esp32/2.1.0/libraries/SD
SPI               2.0.0   /home/[MY_USER]/.arduino15/packages/m5stack/hardware/esp32/2.1.0/libraries/SPI

Used platform Version Path
m5stack:esp32 2.1.0   /home/[MY_USER]/.arduino15/packages/m5stack/hardware/esp32/2.1.0
Error during build: exit status 1
```

Thanks for helping.

arm-ARMY, 2024-02-18:
@sukys read the DEAUTH Prerequisites folder. In order to work you have to make some changes in your arduino installation. There's even a script to make this changes.

n0xa, 2024-02-23:
`multiple definition` error will be fixed if you follow the deauth prerequisites instructions. I'll try to incorporate those into the core README as well.

closed 2024-02-23 by n0xa

sukys, 2024-02-23:
It worked just fine, either for cardputer amt M5Stick. 
As advice, this info could be added to FAQ questions in Read.me

Thanks!