n0xa/m5stick-nemo #103: v2.5.0 wifi captive portal on m5stckc_plus does not redirect to "login url"

lmichoux, 2024-02-18:
I double checked everything. tag v2.4.1 OK   tag 2.5.0 NOTOK
the problem is that somehow when switching to the captive wifi, the automatic login page does not pop up

I am using windows10.
#define STICK_C_PLUS
#define STICK_C_PLUS

also, I tested manual update for DEAUTH as well as install-prerequisites.bat
file is updated correctly.

I swiched back to v2.4.1 and all is ok.

bmorcelli, 2024-02-21:
Hi, there were a misplaced curly brackets at portal_loop function.. I've just opened a pull request to fix this.

lmichoux, 2024-02-23:
> Hi, there were a misplaced curly brackets at portal_loop function.. I've just opened a pull request to fix this.

Great more eyes were needed, I checked the diff but I did not see it. Great job!

n0xa, 2024-02-23:
Fixed in main #107

closed 2024-02-23 by n0xa