n0xa/m5stick-nemo #106: Alternative to an Evil Portal

Rol3oT, 2024-02-20:
Evil Portal is really cool and all, but what if you wanted to capture a password for the Wi-Fi network itself?
Would it be possible to clone a Target SSID and set a very difficult password so that when the user attempts to sign in to their WiFi SSID with the password they know, it will fail, but capture the attempted password?
If there is not a package already in development I was thinking of looking in to making one.

n0xa, 2024-02-20:
Wi-Fi authenticates via a relatively complicated cryptographic process. The vast majority of networks today use WPA or WPA2 with Pre Shared Keys (PSK), with WPA3 on the horizon that will further secure various parts of the Wi-Fi stack. 

There's no straight-forward way to have a user try to authenticate to a rogue access point and expose the actual WPA/WPA2 pre-shared key. You can read up on [how WPA2 cracking works with hashcat](https://hashcat.net/wiki/doku.php?id=cracking_wpawpa2). Tools like Hash Monster, Pwnigotchi, Wifite, and bettercap automate some parts of these attacks, but some of which will illegally send deauthentication attacks against networks you aren't authorized to use, so be careful with them. I've had good luck using hcxtools, hcxpcapng and hashcat together on a Linux laptop, but there are ways to aid data collection with esp32.

I don't foresee NEMO delving too deep into real "pentesting" tool turf. Real, professional pentesters use laptops and best-in-class software for security assessments. Most of the time, they are commandline or text-based specialty tools. I have yet to see a real pro walk up to an engagement and pull out a flipper with a wifi board, a DSTIKE deauthenticator or some other microcontroller gadget, use it for the assessment, and expect to be taken seriously.

Rol3oT, 2024-02-20:
I was more thinking of this
[How can I capture the passwords that people try out on my router (the wrong connection attempts)?](https://www.quora.com/How-can-I-capture-the-passwords-that-people-try-out-on-my-router-the-wrong-connection-attempts)

Exempt the router AP is created by Nemo.

closed 2024-02-20 by Rol3oT