n0xa/m5stick-nemo #109: Error when checking through the Arduino IDE.

05007k, 2024-02-23:
c:\Users\<...>\Documents\Arduino\libraries\1/SD.h:4:18: error: #include nested too deeply
   #include <FS.h>
                  ^
c:\Users\<...>\Documents\Arduino\libraries\1/SD.h:5:18: error: #include nested too deeply
   #include <SD.h>
                  ^
c:\Users\<...>\Documents\Arduino\libraries\1/SD.h:6:19: error: #include nested too deeply
   #include <SPI.h>
                   ^

danny8972, 2024-03-04:
I had this same issue are you sure that in your additional boards manager you have this link 
https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/arduino/package_m5stack_index.json
you can find additional boards manager in Settings for Arduino.

05007k, 2024-03-16:
Thanks for the reply. Yes. I'm sure. I can install the M5 examples with the Arduino IDE.

05007k, 2024-03-16:
Did I do the right thing? That I copied all the libraries associated with this program to the folder "C:\Users \<...>\Documents\Arduino\libraries\1"? And they live as text files, with the extension "H"?

danny8972, 2024-03-19:
I would recommend downloading the library’s with arduinos library downloaded. I’m not familiar with windows but I am certain it has this feature as well, and if you downloaded the files as zip folders try putting the main folder of the unzipped zip file into the library file of your sketch books name.

05007k, 2024-03-19:
Unfortunately, after I left a comment, I played with the device and it stopped programming. 
When connected to a computer, the device is identified as ch340 (341), but the program is not recorded, the device does not respond.
Can you help me?

danny8972, 2024-03-19:
have you tried burning a new image through the m5 app, through the app you can also erase past firmware which should help your issue, just make sure to hold down the G0 button on the top as you plug your cardputer in.

05007k, 2024-03-20:
I tried to record a new image through both the M5 application and the arduino. It doesn't work. This is the first time I've heard about the button. Usually, when recording, I didn't clamp anything. We'll have to try.

closed 2024-08-22 by n0xa