n0xa/m5stick-nemo #110: Ideas for new update

crxyox, 2024-02-24:
- Support for more ir codes like projectors ect
- Ir reciver compatibility

Moz1987it, 2024-02-24:
More power to IR led on devices with S3 by setting gpio pin drive power to 3

From documentation of esp32s3

"The default drive strength of GPIO19 ~ 20 is 2’d3 (~40 mA)" and "The default drive strength of other pins is 2’d2 (~20 mA)"

For example on cardputer gpio44 is IR led so it's 20mA. it's not always on it bursts we can have more range like this

esp_err_t gpio_set_drive_capability(gpio_num_t gpio_num,
gpio_drive_cap_t strength)

danny8972, 2024-03-02:
About the bluetooth thing I don't think its a good idea

look at the read me

Things I probably won't merge in:

Bulk wifi deauthentication spamming
Bluetooth spam that potentially disrupts health and fitness trackers, smart watches, etc.

adsflkj, 2024-04-18:
Where did bluetooth come from? I was looking for more ir stuff.

(This is my alt)
-crxoyx

closed 2024-04-19 by n0xa