n0xa/m5stick-nemo #130: a few problems

SzymonDevs, 2024-03-06:
Hi,
2 days ago i updated my firmware on m5stickc+ to 6.0 and here are all the bugs i noticed(for now, ill update here if u find any more)
1. The random wifi spam either doesnt work or takes me back to main menu
2. Pressing back in the rotation menu rotates the screen(to the right i think).
3. The battery percentage is almost always inaccurate or broken, it even showed me "10090%" one time(id suggest to make a status/top bar with it).
4. The ui colors only saves if you pick custom colors, it never saves for me if i pick one of the themes.

That's it for now, ill update if there are any more i find. Id also suggest to make the tv-bgone a progress bar that will show which code it is currently transmitting(f.ex. if there was 200 codes on 10th code there would be a progressbar at 5% and it would say "10/200".

Regards(and sorry if my english is bad, im Polish and 2 lazy to use a translator),
Szymon

bmorcelli, 2024-03-07:
Hi,

1. The random wifi spam either doesnt work or takes me back to main menu
    - It normally doesn't show anything on screen, and android devices take sometime to notice the networks that Nemo creates, apple devices are faster to notice the networks

2. Pressing back in the rotation menu rotates the screen(to the right i think).
     - I Fixed that in [my repo](https://github.com/bmorcelli/m5stick-nemo/releases/tag/2.6.1b), and I'm creating a PR to fix it.

3. The battery percentage is almost always inaccurate or broken, it even showed me "10090%" one time(id suggest to make a status/top bar with it).
      - pending analisis....

4. The ui colors only saves if you pick custom colors, it never saves for me if i pick one of the themes.
      - I Fixed that in [my repo](https://github.com/bmorcelli/m5stick-nemo/releases/tag/2.6.1b), and I'm creating a PR to fix it.

5. Resettings doesnt do anything.
     - Well, it must get back to original configurations, as it is, can you explain more?

SzymonDevs, 2024-03-07:
Hey,
1. It doesnt create any networks.
3. That might be my hardware fault, i had a problem with it some time ago when if was glitching and rtc clock was just frozen.
5. It doesnt reset the brightness, colors and that is like the only configuration there is to revert

bmorcelli, 2024-03-08:
The "Reset" option is there just to reboot/restart the device, nothing else..

To clear configuration the option is "Clear Settings"

SzymonDevs, 2024-03-08:
By saying reset i meant the "Clear Configuration" option or pressing power+next button while off.

n0xa, 2024-03-10:
I also have a fix for the colors. It's kind of involved and I was still testing it last weekend. I'm in the middle of a cross-country move and won't really be able to approve PRs or patch anything until March 22nd or so. Sorry about that.

closed 2024-08-22 by n0xa