n0xa/m5stick-nemo #153: Bluetooth attacks not working

pillowyfish98, 2024-03-31:
I recently received an m5stickc plus 2 and I flashed the Nemo firmware on it with m5burner I cannot get the Bluetooth attacks working the wifi and ir work but just not Bluetooth I don't know if it is a me issue or something beyond my control but any help is appreciated

Rol3oT, 2024-04-01:
Bluetooth attacks as in bluetooth pairing spam? Which phone are you testing that on? I have found that Android phones are pretty good at blocking current implementation, however changing bluetooth MAC address between sending pairing request (there is a pull request in works) should fix that. I have found that bluetooth spam works really well on IPhones, specially SourApple.

pillowyfish98, 2024-04-01:
I was trying on a Samsung s20 fe. I'll try on an iPhone and see how well that works. Thanks

closed 2024-04-01 by pillowyfish98