n0xa/m5stick-nemo #158: Add Internet Hotspot functionability

swordeye2, 2024-04-09:
Hi!
Is it possible that to make NEMO have internet hotspot capability as well? It would be functional aside from being a very intelligent device :D please.

harrytrinh9, 2024-04-09:
I think it's possible by repeater mode.

n0xa, 2024-04-19:
If someone wants to take a run at this, it would be a good pull request project. I don't have time for it, as it's pretty complex. You will need to:
1) Allocate EEPROM bytes for the SSID and PSK of the access point
2) Add a new menu option in settings 
3) Create a way to update the SSID/PSK with only the buttons on the M5Stick or add an admin panel within NEMO Portal perhaps?
4) Figure out how the Repeater Mode works on ESP32 (this is the first time I've heard of this mode, and I don't know anything about it). it looks like the "esp32_nat_router" project contains many of the first-principles you'd need.

closed 2024-08-22 by n0xa