n0xa/m5stick-nemo #167: Error when compiling in Arduino IDE

tahagedi, 2024-04-26:
I cant compile the file in Arduino IDE, I included all libraries and everything shown in the installation guide. I get a error telling me that I have to include <SPI.h> and when I include it it says that its too deeply nested

danny8972, 2024-04-28:
I had the same issue. I think I fixed it by including the url in settings for additional boards.

davicastaldini, 2024-05-15:
> I had the same issue. I think I fixed it by including the url in settings for additional boards.

How did you do it?

danny8972, 2024-05-15:
Install the M5Stack boards for Arduino IDE: In File -> Preferences, paste this URL into the "Boards Manager URLs" text box. Use commas between URLs if there are already URLs present. https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/arduino/package_m5stack_index.json

closed 2024-08-22 by n0xa