n0xa/m5stick-nemo #173: Portals

eddyizm, 2024-05-06:
Is there a way to add portals, say from evil portals ?

danny8972, 2024-05-06:
Evil portal is already added if you scan and then select a Wi-Fi network you can clone and deauth it

eddyizm, 2024-05-07:
ok, sweet. still waiting on the hardware to come in, a friend demo'd the google captive portal and didn't see other options in my brief look through the code.

danny8972, 2024-05-07:
Glad to help

eddyizm, 2024-05-07:
@danny8972 So it looks like this repo only has one portal, what I was looking for the ability to throw up a different portal like the [pineapple](https://github.com/kleo/evilportals) device. When I get mine I'll fork it and try to add a new one as the html is trivial.

danny8972, 2024-05-08:
Ah sorry for the confusion I see what you mean now. That would be a great thing to add.

pinhoone, 2024-05-09:
The possibility to select a portal would be amazing, I already use this function in my Flipper, do you know a way to set up my custom portal in Nemo?

eddyizm, 2024-05-10:
@pinhoone looking through the code it looks like we would need at least 2 steps. 
1.  a menu item for selecting a portal as opposed to going to the default one and 
2.  adding the html/css/etc in the code for each additional one. This one we should be able to leverage other portal repos. 

Will give this a try when I get my hardware. Waiting on delivery :-)

closed 2024-08-22 by n0xa