n0xa/m5stick-nemo #175: Chip spport and some suggestions

cinouw, 2024-05-15:
Hi there thanks for the great project it's a really good project for the hobbyists like us bu it need some minor adjustments and additions to make it the ultimate pen-testing binary fir the cardputer

1-pn532 NFC/RFID chip spport with this chip you can also emulate NFC/RFID tags or cards and its fairly available 

2-nrf24l01 it's using this we can take control of logitech wireless dongle or any 2.4 ghz dongle with security exploit so nrf24l01 spport would be a nice addition (this chip uses spi interface to connect so internal modifications is needed to make it work bu it is possible)

3- cc1101 spport this the same sub-ghz chip used in the flipper zero and it's a fairly capable one so adding spport for it is a good addition for those who want to mess around with sub-ghz things (this chip uses spi interface to connect so internal modifications is needed to make it work bu it is possible)

4-GPS spport (Neo 6m because it's commonly available) it's already implemented in marauder binary so I think it's an easy addition and it will open a whole new use case for the already integrated WIFI and Bluetooth chips things like wardriving or Bluetooth mapping it also would be a nice addition 

Thanks again for the great work i can't wait for new releases o7

danny8972, 2024-05-15:
These are all great suggestion especially the nrf24l01 I haven't heard of that one before.

cinouw, 2024-05-15:
> These are all great suggestion especially the nrf24l01 I haven't heard of that one before.

Thanks for the spport o7

closed 2024-08-22 by n0xa