n0xa/m5stick-nemo #182: compilation takes AGES

kubilay-dev, 2024-05-22:
hey, i set up nemo in arduino ide and i can compile and upload everything works but its taking literal ages to recompile. is there any way to make this faster? when im working on changing something waiting 10min~ just for testing stuff is really tough especially if you really just need to test and play around with stuff.

i already upped the upload speed but its more the compile process.

Pcquifume, 2024-05-23:
hey,
I don't know how to configure Nemo in the arduino ide it doesn't work I don't understand

vs4vijay, 2024-05-27:
Which OS you're using? Please share your platform config. I believe it is windows, and due to windows defender being run on each compiled files, compilation becomes slower. @kubilay-dev @Pcquifume

kubilay-dev, 2024-05-27:
> Which OS you're using? Please share your platform config. I believe it is windows, and due to windows defender being run on each compiled files, compilation becomes slower. @kubilay-dev @Pcquifume

i have windows defender completely disabled with defender control. and i have no additional anti virus software installed

Pcquifume, 2024-05-27:
Yeah I use Windows.

Pcquifume, 2024-05-27:
but for me he doesn't even want to compile it

closed 2024-08-22 by n0xa