n0xa/m5stick-nemo #184: SourApple Crash stops working after initial success.

makesl0w, 2024-05-29:
Hello NEMO squad!

I have noticed that SourApple Crash only works initially when executed on my M5StickC Plus 2. I am runing a iphone with the latest security update for ios 17 as target. Maybe its the implemented security fix from apple thats causing this, but I think it should be investigated further. Im new to embedded development so it will take some time for me to go through all the code, might need some help @n0xa.

StingerFingersinmyass, 2024-05-30:
Bro dafuq, sourapple crash has fixed like 6 months ago ☠️💀

makesl0w, 2024-05-30:
> Bro dafuq, sourapple crash has fixed like 6 months ago ☠️💀

I know that Apple made a security update to ios preventing it from crashing, however im referring to the spam functionality of SourApple Crash which still works (at least initially).

StingerFingersinmyass, 2024-05-30:
I think I know what you mean, I suggest you contact the Flipper Zero community. Momentum Firmware, Xtreme and etc. . There you will be both helped and noticed.

closed 2024-08-22 by n0xa