n0xa/m5stick-nemo #191: Does it work on m5stickc plus 2 ?

ViBa-Spec0, 2024-07-06:
Recently bought a new m5stickc plus 2 and I was just wondering if Nemo works on it

Dlazder, 2024-07-11:
Yes, it works. But I used the m5 burner for this.

djcasper1975, 2024-07-31:
> Yes, it works. But I used the m5 burner for this.
> 
> Updated: I just managed to download this firmware via the arduino ide, although for this I had to remove the deauth function, as it caused an error during compilation.

I have same issue.

Dlazder, 2024-07-31:
> > Yes, it works. But I used the m5 burner for this.
> > 
> > Updated: I just managed to download this firmware via the arduino ide, although for this I had to remove the deauth function, as it caused an error during compilation.
> 
> I have same issue.

Run the script to install dependencies, it is in the folder in the repository sources. After reboot your PC.

closed 2024-08-22 by n0xa